SynDaverCO/symple-slicer #68: Save option default

synwest, 2020-06-11:
**Save option default**
After loading in an stl and go through the slicing process, the default save option has the file name as x.stl
Note: when the file type is changed to .gcode it stays .gcode
**Details**
 - SympleSlicer Version: v1.0.4
 - Browser [e.g. chrome, safari]
App
**To Reproduce**
Steps to reproduce the behavior:
1. Load in .stl
2. Run through slicing
3. Before clicking save, check save option file type

**Expected behavior**
I believe the save as file type should default to .gcode

marciot, 2020-06-15:
Was the STL extension uppercase, by any chance? It seems like there was a bug where uppercase extensions were not being changed to .gcode

synwest, 2020-06-15:
it is in fact uppercase

marciot, 2020-06-15:
Fixed in v1.0.6, currently updated in beta:

https://syndaverco.github.io/slicer-beta/

synwest, 2020-06-16:
Looks fixed

closed 2020-06-16 by synwest