SynDaverCO/symple-slicer #69: Multiple STL files cause problem with gcode UI

zipsonic, 2020-06-13:
Version 1.0.5 Latest Stable Chrome Windows 10 64Bit

When placing multiple objects, saving gcode results in the extension of the saved file to be STL.

Steps to Reproduce:
1. In Place object, select STL from "Choose File" Dialog. 
2. Click "X" next to STL file listed as there is no option to add another STL. Loaded STL stays on screen.
3. Select another STL in "Choose File" dialog. New part gets autoplaced on bed and moves other STL.
4. Slice Object.
5. Save gcode file.

What should happen:
gcode file autosaves as a .gcode extension

zipsonic, 2020-06-13:
This appears to be closely related to #68.

marciot, 2020-06-15:
Was the STL extension uppercase, by any chance? It seems like there was a bug where uppercase extensions were not being changed to .gcode

zipsonic, 2020-06-15:
I just checked and yes, the file extensions were capitalized. Sorry for the
added work. This can be closed.

marciot, 2020-06-15:
@zipsonic: No need to close until the fix is made. This actually was a bug, just a subtle one :) Thanks for finding it!

marciot, 2020-06-15:
Fixed in v1.0.6, currently updated in beta:

https://syndaverco.github.io/slicer-beta/

SynWillinger, 2020-06-22:
Tested in 1.0.6 and appears to be working for me. Closing issue.

closed 2020-06-22 by SynWillinger