SynDaverCO/symple-slicer #70: Loading multiple STL files - UI problems

zipsonic, 2020-06-13:
Version 1.0.5 Latest Stable Chrome Windows 10 64Bit

The loading of multiple STL files is difficult and poorly executed

Steps to Reproduce:
1. In Place object, select STL from "Choose File" Dialog. 
2. Click "X" next to STL file listed as there is no option to add another STL. Loaded STL stays on screen.
3. Select another STL in "Choose File" dialog. New part gets autoplaced on bed and moves other STL.

What should happen:
Either allow multiple files to be loaded from the Choose File Dialog, or have a place more objects dialog box.

marciot, 2020-06-15:
Good idea! I'll add the ability to select multiple files from the dialog box. FYI, it is now already possible to drag-and-drop multiple files (and this can happen anywhere on the window, not just the drop box)

marciot, 2020-06-15:
Fixed in v1.0.6, currently updated in beta:

https://syndaverco.github.io/slicer-beta/

zipsonic, 2020-06-17:
Looks good on my end. Thanks!

closed 2020-06-22 by SynWillinger