SynDaverCO/symple-slicer #72: Unnecessary Movement in Starting G-Code

zipsonic, 2020-06-16:
`G28                                          ; home all axes
M117 AXI Heating Up...
G29 L1                                       ; load leveling matrix slot
G29 J                                        ; 3 point tilt leveling`

G28 moves the tool to a midpoint of the bed, then raises up 15mm after homing. Then next G0 command is unnecessary because the head has already moved away from the endstops. The resulting movement looks weird, sounds weird and is unnecessary. It also moves the head farther from the origin because G29 J moves it right back closer to the origin for the mesh tilt extrapolation.

Expected Results:

Remove the line

G0 X145 Y187 Z10 F3000                       ; move away from endstops

Because G28 already does that.

synsuiter, 2020-06-16:
This has been applied in testing and will be rolled out to the Symple Slicer shortly. Good catch.

synsuiter, 2020-06-22:
start gcode correction pushed to the machine settings in Symple Slicer.

closed 2020-06-22 by synsuiter