SynDaverCO/symple-slicer #73: Percentage Complete on Axi Does not Update

zipsonic, 2020-06-16:
Ver 1.0.6 - Latest Stable Chrome - Windows 10 64bit

Percentage Complete does not update on Axi Printer.

Starting gcode clears percentage complete with "M73 P0", however, the gcode created by the slicer does not create M73 updates throughout the gcode to update the percentage complete.

Steps to recreate:

1. Load STL
2. Save gcode file
3. Run gcode file

Expected behavior:

Percentage Complete Updates. SympleSlicer needs to be updated to populate the M73 throughout the gcode file.

marciot, 2020-06-16:
I've uploaded a fix to the beta version:

https://syndaverco.github.io/slicer-beta/

zipsonic, 2020-06-17:
Verified Working. Thanks for the quick fix!

closed 2020-06-22 by SynWillinger