SynDaverCO/symple-slicer #74: Bed does not turn off after print

zipsonic, 2020-06-16:
Ver 1.0.6 - Latest Stable Chrome - Windows 10 64bit

Bed does not turn off after print. Came back to my printer hours later with the bed still on.

Ending gcode contains:

M140 S{material_keep_part_removal_temperature_t}   ; keep temperature or cool down

The default slicer profile has this value populated. This value should default to 0, and be changed by the end user as necessary.

zipsonic, 2020-06-17:
This problem still exists in latest updates.

I did some more testing. The ending gcode template contains these lines:

M190 R{material_part_removal_temperature}          ; wait for bed to cool down to removal temp
.
.
M140 S{material_keep_part_removal_temperature_t}   ; keep temperature or cool down

The {material_part_removal_temperature} corresponds to "Slice Objects -> Temperatures -> Part Removal Temperature". By default it's set to 45 degress. I thought maybe {material_keep_part_removal_temperature_t} was a dependent variable like: 
material_keep_part_removal_temperature_t = material_part_removal_temperature + 5

However, changing the material_part_removal_temperature to 0, still results in a gcode file with 

M140 S50 

at the end. I can't find a 50 relating to temperature anywhere in the slicer settings.

marciot, 2020-06-22:
This should be corrected in v1.0.7, now in the beta version.

closed 2020-08-26 by synsuiter