SynDaverCO/symple-slicer #75: Chrome not showing setting windows

Rosey1855, 2020-06-19:
**Describe the bug**
When I refresh the slicer in google chrome I get an error that says slicer is not defined. 

**Details**
 - SympleSlicer Version: 1.0.6.
 - Browser: Chrome

**To Reproduce**
Steps to reproduce the behavior:
1. Go to chrome
2. open slicer
3. Refresh page
4. See error and no setting options

**Expected behavior**
Should refresh to latest chrome version and be able to adjust settings

**Screenshots**

[image]

SynWillinger, 2020-06-19:
I can confirm I am not seeing this on beta or release

Rosey1855, 2020-06-19:
Cleared cache and my issue went away.

closed 2020-06-19 by Rosey1855

marciot, 2020-06-22:
I accidentally pushed a broken version of the updating tool to the beta branch. This hosed things in such a way that clearing the cache was the only way to restore things.