SynDaverCO/symple-slicer #76: Update Error

synsuiter, 2020-06-19:
**Describe the bug**
Browser skips update and does not prompt a second update window. 

**Details**
 - SympleSlicer Version: from version 1.0.6
 - Chrome

**To Reproduce**
I refreshed my page to prompt an update. The Update window popped up and I accidentally hit refresh again before clicking "Update.
The PLA profile that was updated was NOT updated on my browser after that and Samantha's browser WAS updated. I was not able to prompt a new update window top pop up with subsequent refreshes.

**Expected behavior**
If the Update was not successful, the "update" window should continue prompting until it takes. (?)

synsuiter, 2020-06-19:
I imagine I could temporarily fix this by clearing my browsing history, that is the plan for me but idk about customers having to do that.

Rosey1855, 2020-06-19:
@synsuiter is this related to my issue 75?

synsuiter, 2020-06-19:
@Rosey1855 I dont think so but @marciot would have a better idea. I was able to see the settings the whole time I was trying to refresh to prompt an update that wouldn't come.

marciot, 2020-06-22:
If the update mechanism itself gets broken, then it becomes impossible to push out new updates. At that point, the only solution is to clear the cache.

marciot, 2020-06-22:
Before you clear your cache, open up your dev tools and save out any errors for me to see. Thanks.

synsuiter, 2020-06-22:
If we see it happen again we will for sure but we both cleared it Friday before leaving so we could get it working again. I think for now we can add something to the guide to explain to clear browsing data to correct any update/load errors, or maybe we recommend they do it in intervals? I am not sure what the best temporary solution is.

marciot, 2020-06-22:
We are using the beta branch internally and occasionally things will break because of bugs in the code. Customers should not see such bugs (knock on wood) so I would rather see whether it affects customers first before we start freaking out about solutions.

synsuiter, 2020-06-22:
Yeah, but we know to expect an update, they wont automatically know there is a difference to be expected. How would they know their Symple Slicer didn't update? I think a short mention of "a regular clearing of browsing data can ensure you are receiving all updates" or something can help them before they have the problem but don't know they are having it.

marciot, 2020-06-22:
I guess I'll go with whatever you guys decide *shrug*

synsuiter, 2020-06-22:
I was genuinely asking, @marciot How would a customer know they they aren't getting updates? could they check the git hub? can we provide them a link so they can make sure they are on the latest version?

marciot, 2020-06-22:
They could download the source code and check the version number in the "service-worker.js" file. But I know that is not really a solution.

SynWillinger, 2020-06-22:
This is not related to #75. I had this happen as well on Friday. IIRC I simply misclicked when refreshing to update the public version and the update window went away and did not update. I was not able to get it to come back. It seems very feasible a customer will have this problem, especially since we always have a splash screen they have to click past, so they will probably get in a habit of just clicking somewhere every time they open it, which in this case would dismiss the update box and prevent updating.

marciot, 2020-06-22:
@SymWillinger: Merely refreshing the window without clicking "Update" will not skip the update. It will either already be updated by the next time the user visits, or they will be prompted again. I believe something else went wrong.

SynWillinger, 2020-06-22:
I can't really test any further without updates, so we may have to simulate some updates on the beta for testing to be sure on this.

marciot, 2020-06-22:
Okay. The version number in the About screen will stay the same at v1.0.6, but I updated an internal release number to force the update to happen again.

marciot, 2020-06-22:
> I was genuinely asking, @marciot How would a customer know they they aren't getting updates? could they check the git hub? can we provide them a link so they can make sure they are on the latest version?

@synsuiter: With version 1.0.7, the user can bring up both the about box and the change log using the help button. This will allow them to check whether they are falling behind in versions.

synsuiter, 2020-08-26:
@marciot I am going to leave this open in case we get any issues related to it in the field but I think this is a solid solution for the issue.

closed 2021-04-09 by synsuiter