SynDaverCO/symple-slicer #77: Version number not showing on fresh load

SynWillinger, 2020-06-19:
**Describe the bug**
When Symple Slicer is first loaded after a full browser history clear, it is not showing the version number on the splash screen. Was showing the number when memory was not cleared and it had been updated through the updater. 

**Details**
 - SympleSlicer Version: 1.0.6
 - Browser: Chrome

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Clear browsing data"
2. Click on all check boxes in both Basic and Advanced and then "clear data"
3. Open Symple Slicer
4. See error

**Expected behavior**
Version number should be displayed, so we can be sure what version a user is using.

**Screenshots**
[image]

**Additional context**

marciot, 2020-06-22:
This is a known issue. It has been this way for a while. There are no easy solutions but I have been thinking about it.

marciot, 2020-06-22:
Just to provide more context, the version number pertains to the version stored in cache. During the initial load, it is not using a cached version, hence no version number appears. I am thinking that the way around this problem is to present the user with an "installing" screen the first time they load the page, then force a refresh into the cached version. This would avoid this anomalous state where they are using an uncached version.

marciot, 2020-06-22:
It will now always show the update dialog box on first run, which will force the version number to appear when the user clicks "Update!"

marciot, 2020-08-07:
@SynWillinger: Can this ticket be closed?

closed 2020-08-07 by SynWillinger