SynDaverCO/symple-slicer #78: Slight placement difference between browsers

SynWillinger, 2020-06-19:
**Describe the bug**
When generating gcode of the same model from both Chrome and Edge, the placement is ever so slightly different.

**Details**
 - SympleSlicer Version: 1.0.6
 - Browser: Chrome, and Edge

**To Reproduce**
Steps to reproduce the behavior:
1. Import model in chrome
2. Slice file and save
3. Import model in Edge
4. Slice file and save
5. Diff files

**Expected behavior**
Identical files should be generated

**Screenshots**
[image]

**Additional context**
[Chrome_3DBenchy.gcode.txt](https://github.com/SynDaverCO/symple-slicer/files/4806964/Chrome_3DBenchy.gcode.txt)
[Edge_3DBenchy.gcode.txt](https://github.com/SynDaverCO/symple-slicer/files/4806963/Edge_3DBenchy.gcode.txt)

SynWillinger, 2020-06-22:
My work machine received the new Chromium based version of Edge over the weekend, so I decided to test this again with the new version. Weirdly, this still seems to happen. On both browsers I refreshed the page, selected the machine and material profiles to ensure that was the same, loaded the same model and made no changes besides the file name post-slicing but before downloading the gcode.

20mm cube:
[image]

SynWillinger, 2020-06-22:
I tried resetting browsing history on both browsers. Both wound up with different positions from each other, which were also different positions from before.

I then tried a couple tests using just Chrome:
I re-sliced the file after downloading the gcode but without changing anything. This resulted in identical gcode output as expected.
I also tested refreshing the page and then repeating all the same steps I had done before for loading the model and slicing it. This resulted in a location shift like we have been seeing here.

It would appear what we are seeing here is a small lack of repeatability in the placing of the model in the center as it is loaded, unrelated to browser.

marciot, 2020-06-22:
Ohhh. Right. This actually is done intentionally. In order for the circle packing algorithm (which is used for placing objects in a non-intersecting way) it is necessary that the objects not all be in the same position. So I added a small random fudge when placing objects to keep them from overlapping at the origin.

SynWillinger, 2020-06-22:
Ah. It is a little odd to have slightly varying positions when only one model is loaded. Could we have that not activate until the second model is loaded, or is it not worth the hassle and should this be a wontfix?

marciot, 2020-06-22:
Leave it open for now. I have an idea on how to make it better.

marciot, 2020-06-22:
This has been corrected in v1.0.7

SynWillinger, 2020-06-25:
Tested, looks good. Closing issue.

closed 2020-06-25 by SynWillinger