SynDaverCO/symple-slicer #79: Multiple "help" options with different information

SynWillinger, 2020-06-22:
**Is your feature request related to a problem? Please describe.**
There are two places on the main screen where you can see the words "help". One of these is a button at the top right of the screen that opens the guide in a separate window. The other location is in the drop down menu for selecting task groups on the bottom right. The one in the bottom right simply shows what the 3D view mouse controls are. Having these two separate sections with the same label but different information can be confusing to users.

**Describe the solution you'd like**
A link to the guide should be added to the results from selecting "Help" from the drop down in the lower right, so that if they use that instead of the help button at the top right they can still find the guide.

**Describe alternatives you've considered**
Changing the Help button at the top right to say "Guide" instead of "Help".

**Additional context**
[image]

marciot, 2020-06-22:
This has been corrected in v1.0.7

SynWillinger, 2020-06-25:
I like the new functionality. Closing issue

closed 2020-06-25 by SynWillinger