SynDaverCO/symple-slicer #83: UI support settings addition request

synsuiter, 2020-06-30:
**Is your feature request related to a problem? Please describe.**
Based on the variation of object size and geometry, one set of settings for support will be inadequate for many customers. 

**Describe the solution you'd like**
I would like the add these support settings, that can be edited to improve support specifics for a model. 
**Speed for Supports**
**Support Z Distance (OR) Support Top Distance** (I am not sure which overrides which. When applied, this one UI settings should apply to both of these cura settings, I think)
**Support XY Distance**
**Support XY Distance Overhang**
**Support Interface Skip Height**

**Describe alternatives you've considered**
I tried many different settings for supports and while I got a set that works AMAZING for most stuff, it doesn't work perfect for all models. Having these settings in the UI, to allow the customer to vary the distance the support is away or close to the model, would be beneficial. 

**Additional context**
👍

marciot, 2020-07-01:
Pushed fix for this to beta v1.0.7

marciot, 2020-08-07:
@synsuiter: Can this ticket be closed?

closed 2020-08-07 by synsuiter