SynDaverCO/symple-slicer #84: Other UI settings Addition request

synsuiter, 2020-06-30:
**Is your feature request related to a problem? Please describe.**
Adhesion can be effected by the initial layer height and since we do not offer different quality profiles, the customer will need to be able to adjust this settings as it compares to their other layer height changes.
Infill before walls is a settings that can help with thicker layered overhangs, it is a commonly used settings in the 3D printing community. 

**Describe the solution you'd like**
Please add:
**Infill before Walls** as a check box in the SHELL section
**Initial Layer Height** as value entry next to the layer height value

**Describe alternatives you've considered**
NO alternatives, these are being added to prevent an issue that could happen to a customer. 

**Additional context**
Gracias! 😃

marciot, 2020-06-30:
Pushed this to beta.

marciot, 2020-08-07:
@synsuiter: Can this ticket be closed?

marciot, 2020-09-10:
@synsuiter: Can this be closed?

synsuiter, 2020-09-10:
Sorry for the delay, looks good.

closed 2020-09-10 by synsuiter