SynDaverCO/symple-slicer #101: Import Setting being ignored

synsuiter, 2020-08-03:
**Describe the bug**
When I import a profile for testing, my Infill pattern does not apply to Symple Slicer and is not changed with my import. 

**Details**
 - SympleSlicer Version: beta 1.0.8
 - Browser [e.g. chrome, safari]: Chrome

**To Reproduce**
Import a profile with settings "infill_pattern = "cubicsubdiv"" and see that the slice settings still read "lines" and the print produces infill as Lines. 

**Expected behavior**
Infill Pattern to be overridden and read the appropriate pattern when imported. 

**Screenshots**

[image]

marciot, 2020-08-04:
Can you upload the profile you are trying to import?

synsuiter, 2020-08-04:
I cant upload the .toml but I converted it to .txt here:

[ABS-0.5noz_PRODUCTION-PROFILE_BASE-003.txt](https://github.com/SynDaverCO/symple-slicer/files/5022720/ABS-0.5noz_PRODUCTION-PROFILE_BASE-003.txt)

marciot, 2020-08-05:
I am able to confirm it. Fixing this will require investigation and significant work. It will need to wait until the 1.0.9 release after all the desktop stuff is rolled in.

marciot, 2020-08-05:
The reason this is happening is an interaction between "infill_sparse_density" and "infill_pattern". The formula for "infill_pattern" is:

'lines' if infill_sparse_density > 25 else 'grid'

So if you set both "infill_sparse_density" and "infill_pattern" in your profile, then "infill_pattern" will be forced to be "lines", in accordance to the formula.

I'm not sure what the correct behavior is for when the profile specifies things that contradict the formulas

marciot, 2020-08-05:
BTW, here is a trick for you. If you click the three dots in Chrome, then select "More Tools" and "Developer Tools". Then type the following incarnation in the Console:

```slicer.config.verbose=true;```

Then you when you load a profile it will print all the settings and how they were computed:

[image]

This may help in understand what is happening in the future.

marciot, 2020-08-05:
I've confirmed this bug has already been fixed in the latest code. The reason the fix isn't in the beta is because we backtracked to 1.0.7, which lacks some of the improvements in the 1.0.8 code we abandoned (and which will now be present in 1.0.9)

marciot, 2020-08-05:
Nevermind previous comment. There was another bug. Let me see what I can do.

marciot, 2020-08-05:
Okay, I pushed a fix to this to the beta.

marciot, 2020-08-07:
@synsuiter: Can this ticket be closed?

closed 2020-08-07 by synsuiter