SynDaverCO/symple-slicer #102: TPU 95 "bridging" settings

synsuiter, 2020-08-05:
**Describe the bug**
When I export the TPU 95 Profile I get different Bridging Speed settings than in the direct profile .toml file. 

**Details**
 - SympleSlicer Version: 1.0.8
 - Browser [e.g. chrome, safari] ; Chrome

**To Reproduce**
Steps to reproduce the behavior:
Apply Syndaver Axi Machine and TPU Shore hardness 95 profile. then g and export that profile. Confirm by comparing to the git hub direct profile file. 

**Expected behavior**
To export the settings that were set initially. 

**Screenshots**
[image]

synsuiter, 2020-08-05:
Also Material_final_print_temp and Material_initial_print_temp settings exporting incorrectly. 

[image]

marciot, 2020-08-05:
Have you refreshed since I put in a fix for #101? Probably same thing.

synsuiter, 2020-08-05:
I thought I had refreshed but it just updated for me, looks good.

closed 2020-08-05 by synsuiter