SynDaverCO/symple-slicer #104: Adding Parts to the bed

synsuiter, 2020-08-11:
**Describe the bug**
When adding parts to the bed, the 1st 3 have no issue, but addition 4-6 do not appear to have been added, THEY ARE ADDED, but they have been placed over the top of the other previously aided parts. Then when I add #7, they all snap to an area showing them but they are still overlapping. 

**Details**
 - SympleSlicer Version: 1.0.8
 - Browser: Chrome

**To Reproduce**
Steps to reproduce the behavior:
I used the model for our spool arm, this happens more with larger parts. 
All I did was add the parts, the application of machine and profile settings do not seam to make a differece. 
Please see images bellow for process. 

**Expected behavior**
The parts to be added next to one another or off the bed (if they don't fit on it) instead of over the top of one another. 

**Screenshots**
[image]
[image]
[image]
[image]
[image]
[image]
[image]

marciot, 2020-08-11:
Please upload the STL files.

marciot, 2020-08-11:
FYI, the packing algorithm used by Symple Slicer circumscribes all the parts in a circle and then tries to arrange those circles. Thus, it will never be able to pack long skinny parts next to each other. Those parts will always need to be manually adjusted in some way.

marciot, 2020-08-11:
I'm a bit surprised this works for parts 1 through 3. It shouldn't work at all, TBH, because the circumscribed circles are overlapping. In cases like this, maybe Symple Slicer should tell the user automatic placement failed and they need to position the parts manually.

marciot, 2020-08-11:
Arranging parts on a print bed is the sort of thing that is obvious to humans, but computers cannot do very well.

synsuiter, 2020-08-11:
Can't upload the STL model, not supported, but I can sent it to you in an email if you would like. 

Good to know, I figured this was the case. We can use the info in this ticket to explain to customers its limitations in the future.

closed 2020-08-11 by synsuiter