SynDaverCO/symple-slicer #105: Parts shifting after profile application

synsuiter, 2020-08-11:
**Describe the bug**
When I add parts to the bed and get it centered, then apply a profile (either through the menu or an import), the parts then rearrange themselves back to the center, even if I had arranged them specifically or merely had them centered in a different way. This happens with multiple and stand alone parts. 

**Details**
 - SympleSlicer Version: 1. 0.8
 - Browser: Chrome

**To Reproduce**
Steps to reproduce the behavior:
Open Symple Slicer
Add part(s) to bed
Arrange them as desired
Apply your selected machine and print profile (or you can import a desired profile)
Watch the parts center and move around

**Expected behavior**
Part positions should not be adjusted when a profile is applied, they should remain stationary. 

**Screenshots**
[image]
[image]

marciot, 2020-08-11:
@synsuiter: Loading a profile may change the bed dimensions, so right now it triggers the object placement routine. It may make sense to simply have an "automatically arrange bed" checkbox that starts out checked, but gets unchecked any time the user manually positions an object. That way Symple Slicer won't ever mess up what the user has done.

synsuiter, 2020-08-11:
@marciot That seems like a good solution, we can place this on low priority, no direct complaints yet through tech support.

synsuiter, 2021-04-09:
This has not yet been an issue for our customers. I am going to close this out.

closed 2021-04-09 by synsuiter