SynDaverCO/symple-slicer #106: Large Model

synsuiter, 2020-08-11:
**Describe the bug**
After applying a profile and watching the model move off the bed, when it is moved back on the bed, it still says it is unprintable. 

**Details**
 - SympleSlicer Version: 1.0.8
 - Browser: Chrome

**To Reproduce**
-Add large model
-Arrange Model on the bed
-Apply a profile and watch the model shift (this issue is already addressed in #105)
-Center your model and watch that it is still red.

**Expected behavior**
It should register as printable within the build space. 

**Screenshots**
[image]
[image]
[image]

Additional Info:
The only way to fix it is to move the model above the bed and letting it fall back to the build surface, that makes it printable again.

synsuiter, 2020-08-11:
I can send you this model through email but i cant upload in in a comment here.

marciot, 2020-08-12:
Change the file extension to .txt

marciot, 2020-09-18:
@synsuiter: I believe this has been fixed in the latest beta.

synsuiter, 2021-04-09:
This appears to be good to go.

I do want to note that when the model is sized down, it does not "center" on the bed, but is slightly off center. Since this is not a major issue, I am closing this out and will open a new ticket if the centering becomes an issue.

closed 2021-04-09 by synsuiter