SynDaverCO/symple-slicer #107: Mirror tool bug with lithophanes

synsuiter, 2020-08-13:
**Describe the bug**
When using the arrows on the object, the mirror tool does not mirror the object, it stretches it along the axis chosen. 

**Details**
 - SympleSlicer Version: 1.0.8
 - Browser [e.g. chrome, safari]: Chrome

**To Reproduce**
Load in a lithophane
Select the object
Click the Mirror tool
Use the arrows on the object to mirror the part
Watch it stretch
Side Note: when I used the same tool/button/axis a second time, it did mirror the part. 

**Expected behavior**
The part should mirror on the first click with those buttons

**Screenshots**
[image]
[image]
[image]
[image]
[image]
[image]
[image]

marciot, 2020-09-14:
This has been fixed in the beta.

synsuiter, 2020-09-14:
Gracias!

closed 2020-09-14 by synsuiter