SynDaverCO/symple-slicer #108: M73 post-processor toggle

SynWillinger, 2020-08-13:
**Is your feature request related to a problem? Please describe.**
Not all gcode interpreters use the M73 command, which is being added by a post-processor in Symple Slicer currently.

**Describe the solution you'd like**
Per a discussion with @marciot, a toggle based on the "machine_gcode_flavor" profile setting would probably work well for this.

**Describe alternatives you've considered**
Another, per-machine profile toggle.

**Additional context**
This is low-priority for now, but could become more important sometime in the future.

marciot, 2020-11-30:
@SynWillinger: A fix for this is now in beta (v1.0.15), close if it looks good.

SynWillinger, 2020-11-30:
@marciot is this tied to the machine_gcode_flavor setting? If not please let me know where it is. Will test once I know how to test.

SynWillinger, 2020-12-01:
Went ahead and tried changing the machine_gcode_flavor setting and this appears to be working.

closed 2020-12-01 by SynWillinger