SynDaverCO/symple-slicer #109: Profile selection confusion

synsuiter, 2020-08-14:
**Describe the bug**
When you apply a selected material profile, and select "last session settings" for the machine, it uses all your last session settings for the material profile also.

**Details**
 - SympleSlicer Version: 1.0.8
 - Browser [e.g. chrome, safari]: Chrome

**To Reproduce**
Steps to reproduce the behavior:
Go to Symple Slicer
Apply your machine and material profile
Load in a 3D object
Go change some settings in the material profile
Slice
Go apply your selected material profile and leave the Machine profile selection to "Last session settings"
observe that the material settings that were altered are still altered. 

**Expected behavior**
Machine profiles should override machine settings and material profiles should override material settings.

**Screenshots**
[image]
[image]
[image]
[image]

marciot, 2020-08-14:
This should not be happening. Which profiles are you using?

marciot, 2020-08-17:
I think I see what is happening. I tried the following:

- Opened Symple Slicer
- Selected "SynDaver AXI" and "PLA"
- Hit Apply.
- Went to "Shell" in "Slice Objects" and changed "Top/Bottom Pattern" from "Lines" to "Concentric"
- I then sliced and clicked "Do It Again!"
- I was taken to "Select Profiles" and I noticed that the Printer said "SynDaver AXI" and the material said "PLA"
   - This is confusing and probably the reason for the request #110 

- - Anyhow, I changed machine to "keep previous settings" and material to "PLA"

- I then went to "Shell" in "Slice Objects" and looked at "Top/Bottom Pattern" and noticed it said "Concentric"
   - I was a bit confused by this, because I too expected it to be reset to "Lines" but then I looked in the profile and I understood why this was happening.

"Top/Bottom Pattern" (i.e. top_bottom_pattern) is not listed in the either the material nor the printer profile. Therefore, it will maintain whatever value the user changed it to. It will never be reset by a profile -- because it is not part of a profile.

Now I understand why this came about. We had a long discussion a while back where we decided we were going to simplify our profiles and keep Cura defaults. This meant removing a bunch of stuff from the profiles, when the Cura startup defaults were appropriate. This is still a good idea, but it introduces the problem we are now seeing.  Here is what happens when the SympleSlicer "Apply" button is pressed:

1. If "Printer" and "Material" are both something other than "Last session settings"
   * Clear all settings to their Cura defaults (including whatever the user modified)
   * Load all settings in "Printer" profile
   * Load all settings in "Material" profile
2. If "Printer" is "Last session settings" but a "Material" is selected.
   * Keep all settings as they were
   * Load all settings in "Material" profile
3. If "Material" is "Last session settings" but a "Printer" is selected.
   * Keep all settings as they were
   * Load all settings in "Printer" profile

So, what happens if a setting is neither in the printer nor material profile? Then in cases 2 or 3, the setting will maintain its state. It will only get restored to the default in case 1.

So the question is, how do you tell Symple Slicer that a particular setting is a Material setting, but we want it to use the Cura defaults? Currently there is no way to do this. There are two possibilities:

1) We can come up with a new file that will tell Symple Slicer which settings are Material settings and which ones are Printer settings.
2) We can add a new value to the Material profiles called "cura_default" which will tell Symple Slicer that when that profile is selected, we want to restore that setting to the cura default (overriding whatever the user has selected).

marciot, 2020-08-17:
I think there might be a UI change that solves this problem. The UI currently allows you to select either "Printer" or "Material" as "Last session settings" independently.

I propose:

- A radio button labeled "Use last session settings"
- A radio button labeled "Select new profile"
- The two drop downs which will be invisible or grayed out if the first radio button is selected:
  - Material: Default selection SynDaver AXI
  - Printer: Default selection PLA

With this UI change, then Symple Slicer can always load the Cura defaults before loading the Material and Printer profiles, so there will be no ambiguity and we get what we expect.

synsuiter, 2020-08-17:
I like this, breaking it up definitely sounds like the way to go, good idea.

synsuiter, 2020-08-26:
One new issue from this update is that if the second option loads in selected, it does not show the profile selection options. It needs to be un selected and then re selected to make them pop up, we have one instance of someone thinking they didn't know where to find profiles. can we make it so those options are just always visible and do not disappear? I know that may cause confusion when someone is trying to select their previous settings, any ideas or thoughts?

marciot, 2020-08-26:
The second option should never load selected. If this is happening, then it is a bug. Let me know under what circumstances it happens. This may be related to another ticket I already closed.

synsuiter, 2020-08-26:
oh, that is my bad, I was not clear, that's not what is happening. we are trying to repeat and get an image now. What we are seeing is that the drop down options some times do not load in even when the second option is selected, and in order to make those drop down options visible, you have to select the top bubble and then reselect the second bubble to view those options. 
it has happened more than once on at least two computers but we are having issues repeating it now, will let you know how it goes.

marciot, 2020-09-10:
@synsuiter: Can this be closed?

synsuiter, 2020-09-10:
I think we can close this due to our inability to repeat it, we can reopen if we are able to learn/document more.

closed 2020-09-10 by synsuiter