SynDaverCO/symple-slicer #110: Profile "Apply" override warning

synsuiter, 2020-08-14:
**Is your feature request related to a problem? Please describe.**
There is a bug I submitted in relation (#109) but we would like this feature to be added no matter what way the other ticket goes.

**Describe the solution you'd like**
Can we add a warning pop up that will ask the user to verify whether they are sure they want to "Apply changes and override old settings?"? We are concerned that it is not clear enough to a newer user that applying settings will cause everything to be changed and will delete any changes made by the user previously. 

**Describe alternatives you've considered**
We have considered changing the wording of the button or the description next to it. Perhaps adding a "Next" button so the user does not have to use the "apply" button to move on to the next step. (I know they can use the drop down but if they are new, they may no know that yet) But then we might need to add a warning to the "next" button in case someone tries to slice before applying anything. Just not sure on this one. 

**Additional context**
USER EXAMPLE: This was a find by Curt. In his use this was not as clear to him as he would have liked and caused some setting confusion.

marciot, 2020-08-14:
The default option is to keep the settings. so it seems like the user won't lose their settings unless they take the specific action of selecting another profile. Isn't this sufficient to avoid accidental loss?

marciot, 2020-08-17:
Okay. My previous comment is incorrect, the default option is to keep the settings only on page reload. It is not the default option when doing a second print.

Here is what I tried:

- Opened Symple Slicer
- Selected "SynDaver AXI" and "PLA"
- Hit Apply.
- Went to "Shell" in "Slice Objects" and changed "Top/Bottom Pattern" from "Lines" to "Concentric"
- I then sliced and clicked "Do It Again!"
- I was taken to "Select Profiles" and I noticed that the Printer said "SynDaver AXI" and the material said "PLA"
   - This is confusing, because it means the user will change their settings is they hit "Apply" 
   - It is also inconsistent with what happens if they reload the page.

There are two possible solutions:
   - We could change both to "Keep Previous Settings" and "Keep Previous Settings" once the user has sliced something.
   - In that case, I think the warning would not be necessary, because the default option will not destroy user choices.

A more complex solution would try to keep track of whether the user has in fact changed something and only show "Keep Previous Settings" when it made sense to do so, but we need to solve the problem with #109, finding a way to tell Symple Slicer which settings are in fact material settings, which are printer settings, and which are neither. Currently, Symple Slicer does not know this.

marciot, 2020-08-17:
I think there might be a UI change that solves this problem and #109 in one shot. The UI currently allows you to select either "Printer" or "Material" as "Last session settings" independently.

I propose:

A radio button labeled "Use last session settings"
A radio button labeled "Select new profile"
The two drop downs which will be invisible or grayed out if the first radio button is selected:
Material: Default selection SynDaver AXI
Printer: Default selection PLA
With this UI change, the user has to keep the last settings or select both a material and printer. This is a case Symple Slicer can easily handle: If the second radio button is highlighted, Symple Slicer loads the Cura defaults (to clear all user changes), then loads the Material and Printer profiles. Everything will be exactly as we expect because no user changes will be carried over.

marciot, 2020-09-10:
@synsuiter: Can this be closed?

synsuiter, 2020-09-10:
I am going to close this, looks good, if we need any more edits in the future we can open a new request.

closed 2020-09-10 by synsuiter