SynDaverCO/symple-slicer #111: Support interface option

synsuiter, 2020-08-28:
**Is your feature request related to a problem? Please describe.**
The Support settings often need an interface to succeed

**Describe the solution you'd like**
Under the "Enable Support Brim", can we add "Enable Support Interface" to activate the setting "support_interface_enable"?

**Describe alternatives you've considered**
I considered having it active in the background without access through the UI but it uses a lot of material and is only need for complex and organically shaped supports. A UI option would be better. 

**Additional context**

marciot, 2020-08-31:
Please check this in the latest beta.

marciot, 2020-09-10:
@synsuiter: Can this be closed?

synsuiter, 2020-09-10:
Works great! thank you

closed 2020-09-10 by synsuiter