SynDaverCO/symple-slicer #112: Initial Layer Speed

synsuiter, 2020-08-28:
**Is your feature request related to a problem? Please describe.**
lower layers are having a harder time sticking due to higher than usual default settings, control to the customer is needed for various adhesion methods. 

**Describe the solution you'd like**
A UI option for the First layer speed, independent of the "print speed" option. 

**Describe alternatives you've considered**
Slower speeds overall is the only alternative and we have chosen not to reduce speeds overall just for the 1st layers adhesion. 

**Additional context**

marciot, 2020-08-31:
Please check this in the latest beta.

marciot, 2020-09-10:
@synsuiter: Can this be closed?

synsuiter, 2020-09-10:
This is good to go, thank you!

closed 2020-09-10 by synsuiter