SynDaverCO/symple-slicer #113: Infill density changes infill pattern

synwest, 2020-08-31:
**Describe the bug**
When the infill density is changed, the infill pattern id automatically changed to lines instead of staying at what the profile sets it as

**Details**
 - SympleSlicer Version:1.09
 - Browser Chrome desktop

**To Reproduce**
Steps to reproduce the behavior:
1. Load in profile
2. Move into slice object menu
3. Change infill density
4. infill pattern changes to lines

**Expected behavior**
The infill pattern should be unchanged unless specifically changed through the infill pattern dropdown

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

marciot, 2020-08-31:
The reason for this is that there is a formula defined for "infill_pattern":

```
'lines' if infill_sparse_density > 25 else 'grid'
```

We can remove the formula, but then this automatic heuristic would no longer happen.

synwest, 2020-08-31:
@synsuiter

synsuiter, 2020-08-31:
hmm, this is an odd one. I see negative potential for the customer, on this one. The infill pattern can change a lot about the outcome of the part. 
@marciot , What is happening with that formula? Is it always trying to set the pattern to lines if it is  less that 25% and grid if it over 25%? This seems like something we should definitely get rid of since our profiles set the % and the pattern and if the customer changes it, they have no way of knowing its going to change what they wanted if you do it in the wrong order.

What about another option, could we highlight the pattern if the density gets changed? then if the pattern gets changed it becomes UNhighlighted? Maybe not that exactly but something like that?

@marciot and @synwest Let me know your thoughts, this is a tough one I think.

marciot, 2020-08-31:
> hmm, this is an odd one. I see negative potential for the customer, on this one. The infill pattern can change a lot about the outcome of the part.
> @marciot , What is happening with that formula? Is it always trying to set the pattern to lines if it is less that 25% and grid if it over 25%?

Other way around. It will select lines for infill over >25%, grid otherwise. Not sure why. It would seem like they are making the infill weaker when the user is requesting a stronger print. So it kind of goes against what the user desires.

> This seems like something we should definitely get rid of since our profiles set the % and the pattern and if the customer changes it, they have no way of knowing its going to change what they wanted if you do it in the wrong order.

Yeah. I agree.

> What about another option, could we highlight the pattern if the density gets changed? then if the pattern gets changed it becomes UNhighlighted? Maybe not that exactly but something like that?

Adding something like this would be a pretty major endeavor. Certainly won't happen for a while.

> @marciot and @synwest Let me know your thoughts, this is a tough one I think.

Given that the logic behind the formula seems to go against the user's wishes, I think we simply remove the formula.

synsuiter, 2020-08-31:
Agreed. @marcio removing the formula seems like the best option.

closed 2020-09-10 by synwest