SynDaverCO/symple-slicer #114: Gcode naming save window

synsuiter, 2020-09-01:
**Is your feature request related to a problem? Please describe.**
The box where you type the name to save the gcode, is too small. 

**Describe the solution you'd like**
allow a larger base area for this naming box. 

**Describe alternatives you've considered**

**Additional context**

marciot, 2020-09-01:
I personally don't like having that text box in there at all. It is a kludge for a misconfigured browser. If the user wants to have an ability to change a file's name, the correct way to do it is to have the browser ask prior to saving files.

In Chrome, you go under Advanced, Downloads and check "Ask where to save each file before downloading". Once you do this, not only do you get to select a name, but you also get to select the location. This should be the default, IMHO.

Anyhow, the reason the textbox is short is because the field labels are all the same width throughout the UI. This is what keeps the UI visually aligned. It doesn't leave much space for a text entry box. I would have to break the layout rules for that particular text box if we wanted it to look different.

Another alternative would be to keep the box as is, but make it non-editable. Clicking on it would bring up a dialog box like this for them to edit the filename:

https://mdn.mozillademos.org/files/11303/prompt.png

marciot, 2020-09-10:
@synsuiter: Fix for this has been pushed to beta.