SynDaverCO/symple-slicer #115: Nozzle Distance from Bed

CakeEaterMan, 2020-09-05:
When printing a larger layer height the nozzle prints too far above the bed. This causes the initial layer to peel up. 

The ability to adjust the distance of the nozzle from the bed would resolve the issue.

marciot, 2020-09-16:
@synsuiter: Please advise what should be done in this situation and whether any thing needs to be changed in SympleSlicer.

synsuiter, 2020-09-16:
@CakeEaterMan Good point, luckily there is already the ability to adjust the height on the machine through the LCD screen. 
SympleSlicer cannot adjust for your machines nozzle Z offset because it does not know the specific bed leveling calibrations for every machine, it must be adjusted on the individual machine. 

Here is how to adjust while the skirt or other adhesion type is printing:
MENU --> Nudge Nozzle --> then choose your mm adjustment amount (0.01, 0.1, 1) and adjust down or up. Adding will raise the nozzle up, and subtracting will lower the nozzle to the bed. 

Here is how to adjust while there is no print in progress:
MENU --> Advanced Settings --> Probe Z Offset -->then choose your mm adjustment amount (0.01, 0.1, 1) and adjust down or up. Adding will raise the nozzle up, and subtracting will lower the nozzle to the bed. 
**NOTE: if you use this way to adjust your z offset please adjust  by no more than 0.05 mm at a time. Setting too low of a nozzle z offset and then beginning a print can cause your nozzle to dig into the bed and damage your printer.** 

I hope this helps out. :D Have a good one!

CakeEaterMan, 2020-09-16:
@synsuiter Thanks for the tip about manually adjusting the nozzle. I understand why SympleSlicer is unable to adjust for a specific machine, but it is adjusting the nozzle's distance from the bed based on the provided layer height. 

For example:
If I set the layer height in SympleSlicer to 0.2 mm then during the print the print head is 0.2 mm away from the bed, and if I set the layer height in SympleSlicer to 0.5 mm then during the print the print head is 0.5 mm away from the bed. 

There is no problem with the former, but the later is too far from the bed and will cause an issue unless the z offset is manually adjusted.

marciot, 2020-09-17:
@CakeEaterMan: Why can't you change the "Initial Layer Height" to whatever you want it to be?

CakeEaterMan, 2020-09-20:
Because that is not the only thing it affects.

marciot, 2020-09-21:
@CakeEaterMan: Gotcha.

So the only answer I can give you at this point is that SympleSlicer does not have what you are asking for because CuraEngine does not have this feature (AFAIK). SympleSlicer isn't a fork of the slicing engine, it is just a different front-end. If you want a new slicing engine feature, it would need to be requested here:

https://github.com/Ultimaker/CuraEngine

closed 2021-04-09 by synsuiter