SynDaverCO/symple-slicer #116: Parts moved outside of area still red after place in center

SynWillinger, 2020-09-08:
**Describe the bug**
If a part winds up outside of the print area during setup, it will turn red. If the part is then moved back inside the print area using the right click menu "Center selected object" option, it will stay red.

**Details**
 - SympleSlicer Version: 1.0.10
 - Browser: Chrome

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Place Objects"
2. Place an object
3. Select the object
4. Use the "move object" position fields, or click and drag objects, to move at least part of the object outside of the print area
5. with the object still selected, right click and select "Center Selected Object"
6. Observe that the object is still colored red, and in the object manipulation menus it will say "The red objects are outside the printable area."

**Expected behavior**
Part should stop being red and error message should go away once part is back in printable area

**Screenshots**
[image]

**Additional context**
It seems to recognize that it is within the limits and correct itself if moved from the position input boxes or grabbable objects, just not when placed direct back to center from the right click menu.
It will still slice just fine while showing up as red.

marciot, 2020-09-14:
This has been fixed in the beta.

SynWillinger, 2020-09-14:
Tested. Looks good to me.

closed 2020-09-14 by SynWillinger