SynDaverCO/symple-slicer #117: Material - added Info on Selection

synsuiter, 2020-09-15:
**Is your feature request related to a problem? Please describe.**
As customers have a wider range of materials to choose from, we need a method to communicate needed information to them when choosing profiles. For example, when they switch from PLA to Nylon material, they need to know that Nylon is sensitive to colder environments and airflow. Or when they print TPU they will need to put glue stick on the bed so their print is removable.

**Describe the solution you'd like**
I think a pop up or a warning is too negative and would alert the customer to an error. Instead I think we should go with red or highlighted text that appears inside the UI window when selecting which material profile to use. For Example Nylon would have text bellow the selections that reads "Sensitive to airflow and cold environments - Apply PVA glue stick to print bed", PLA night read "Sensitive to warmer environments", and so on. (those are not what I want there, those are only examples)

**Describe alternatives you've considered**
The only alternative that I can think of is that we offer this information in the Filament Guide I am working on but it is going to be a while before I am done and I am concerned about relying on the customer to read the instructions first. Text that just appears when you choose that material, seems like the best option. 

**Additional context**
I have not decided on the specific text for each material. I am working on that.

marciot, 2020-09-16:
This feature has been pushed to beta. Simply add a "material_notes" field to the "[metadata]" section of the material in question, example:

```
[metadata]

date = "8/26/2020"
material_notes = "This is a test"

[settings]

adhesion_type = "skirt"
infill_before_walls = false

...

```

marciot, 2020-10-06:
@synsuiter: Can this be closed?

closed 2020-10-06 by synsuiter