SynDaverCO/symple-slicer #118: First layer % not controlled

SynWillinger, 2020-09-24:
**Describe the bug**
Symple Slicer is controlling the print completion % through M73 commands at each new layer. When not controlled through M73, the marlin firmware will give % based on how many bytes of code have been run. Testing has shown that the first layer will use that second method, but after the second layer it stays with what the gcode tells it. This can result in the % going backwards when moving to the second layer. Testing was done with 16 20mmX20mmX5mm cubes and when it reached the second layer it went from 22% to 20%.

**Details**
 - SympleSlicer Version: 1.0.10
 - Browser: Chrome

**To Reproduce**
Steps to reproduce the behavior:
1. Run a print with a decent sized first layer that isn't very tall
2. Observe the % increase as the first layer progresses
3. Observe the % shift on the first layer change
4. Observe the % stay the same until the next layer change

**Expected behavior**
The % should be controlled by only one method, and should not go down

**Additional context**
The first layer is receiving a M73 command, but the command is 0. Based on what I have observed, setting it to 0 will reset it but allow marlin to take control of it. However, setting it to something besides 0 does not result in marlin taking control. Having the first layer start at 1% may fix this.

marciot, 2020-11-30:
Okay. So this is a two part fix.

The first is that Symple Slicer should never insert a M73 P0. So I looked at the code and made this change.

But while doing so I also realized Cura puts in time estimates at various points in the file, so rather than use layers to compute the progress, I am now using that time estimate divided by the total time to get a better percentage value.

Now the second part of the fix is that for the first layer, the progress is set in the start GCODE. Rather than having a "M73 P0", I changed the start GCODE to use "M73 P1". This, along with Symple Slicer never writing out M73 P0, should fix the retrograde temporal leap.

SynWillinger, 2020-11-30:
Sounds promising

marciot, 2020-11-30:
@SynWillinger: A fix for this is now in beta (v1.0.15), close if it looks good.

SynWillinger, 2020-12-01:
@marciot I still see the gcode as showing M73 P0 in the start gcode. Tested using ABS profile, freshly loaded from Select Profiles panel at start of workflow.

Other than that it appears to be working, but I intend to throw a couple of different cases at it in the morning to confirm.

SynWillinger, 2020-12-01:
Testing complete. Aside from the aforementioned start gcode issue, this looks to be working quite well.

SynWillinger, 2021-04-13:
This has been fully fixed

closed 2021-04-13 by SynWillinger