SynDaverCO/symple-slicer #120: Add layer 0 bed temp setting

synsuiter, 2020-09-30:
**Is your feature request related to a problem? Please describe.**
There has been confusion in house and some frustration in the field about not having the first layer bed temp setting option available because there is no way to indicate to the customer that the 1st layer temp will be different. Adding this setting into the UI will alleviate the confusion about the change and give customers that ability to edit their 1st layer temps for more materials. 

**Describe the solution you'd like**
Add the setting from cura "material_bed_temperature_layer_0" into the UI in Symple Slicer as "First layer bed temperature".

**Describe alternatives you've considered**
The attempt has been made to communicate why this feature is different and not visible but that does not seem to be alleviating the confusion or frustration at not being able to customize it. 

**Additional context**
None.

marciot, 2020-09-30:
So both the material_print_temperature_layer_0 and material_bed_temperature_layer_0 are commented out. Do you just want to bring back material_bed_temperature_layer_0?

marciot, 2020-10-06:
@synsuiter: Can this be closed?

closed 2020-10-06 by synsuiter