SynDaverCO/symple-slicer #121: Add initial print temp to the UI

synsuiter, 2020-10-02:
**Is your feature request related to a problem? Please describe.**
Higher temperature materials that begin with our profiles may need a larger temp increase on the 1st layer for bed adhesion more frequently than we had expected. 

**Describe the solution you'd like**
Add the CURA setting "material_print_temperature_layer_0" to the UI of Symple Slicer. 

**Describe alternatives you've considered**
None. (setting a higher default temp could hinder lower printing temperature materials)

**Additional context**
I have a concern about the setting "material_initial_print_temperature". Which sets the printing temperature for when the 1st layer can begin printing. It is still set to the default and I have a concern that it will begin printing at too low of a temp and then ramping up during that first layer could be VERY inconsistent. Is there a way to tie the two together so it will always wait for the layer 0 temp before beginning?

I feel this will need further discussion.

marciot, 2020-10-05:
Adding the "material_print_temperature_layer_0" to the UI is trivial. I'll do that.

As for your other question, the formula for "material_initial_print_temperature" is "max(-273.15, material_print_temperature - 10)" I have no idea where -273.15 comes from, but we could change it to "max(-273.15, material_print_temperature_layer_0 - 10)" if you think it would be better.

> **Is your feature request related to a problem? Please describe.**
> Higher temperature materials that begin with our profiles may need a larger temp increase on the 1st layer for bed adhesion more [...]

synsuiter, 2020-10-05:
Can we do "material_print_temperature_layer_0 - 5" instead? That would keep things in line with other windows of 5 degrees and link it to the proper temp setting for its use.

marciot, 2020-10-06:
@synsuiter: Can this be closed?

closed 2020-10-06 by synsuiter