SynDaverCO/symple-slicer #122: Center Model with multiple on the bed

synsuiter, 2020-10-05:
**Describe the bug**
When trying to center an object with multiple objects on the build plate, it centers all objects. 

**Details**
 - SympleSlicer Version: Beta 1.0.11
 - Browser [e.g. chrome, safari]:  Chrome

**To Reproduce**
Steps to reproduce the behavior:
Load multiple objects
Move each to desired locations around the bed
select only one model and center it
Observe all other model move to grouped center

**Expected behavior**
Only the selected model should move to center

**Screenshots**
[image]
[image]
[image]

marciot, 2020-10-05:
I cannot fix this. When you ask to center a model, it changes the position on the first object, but then has to run the packing algorithm to make sure the objects don't overlap. This causes all objects to gravitate towards the center of the bed.

marciot, 2020-10-05:
I can ask the developer of the packing library whether there is a way to modify the algorithm so it doesn't try to pack things together, but merely avoid overlaps, but he may or may not be able to do this.

synsuiter, 2020-10-05:
We could consider changing the wording of the tool to be "Center Objects" instead of specifying the selected one. If we cannot fix it, at least we can explain what it does more clearly.

marciot, 2020-10-05:
So I looked at the code for the library. Even though the "packing" part is sort of an essential component for what the library is supposed to do (it's meant to pack circles close together, while we are sort of misusing it to prevent overlaps of parts), it looks like I might be able to turn off that functionality. I'm going to try out the modification and see what happens. It either will do exactly what we want, or sh*t will hit the fan and it won't work at all. We'll see.

synsuiter, 2020-10-08:
I haven't noticed any other issues but the centering is still working the same. I can post images if needed but they will appear the same.

synsuiter, 2020-10-08:
New update is looking good! no errors in centering, we will keep up testing and report back soon.

marciot, 2020-11-18:
@synsuiter: Can this be closed?

synsuiter, 2020-11-23:
Indeed! :D

closed 2020-11-23 by synsuiter