SynDaverCO/symple-slicer #123: Z Seam: "User Specified"

synsuiter, 2020-11-02:
**Is your feature request related to a problem? Please describe.**
If the user selects the Z seam "User Specified" they do not currently have pop up settings allowing them to specify the X and Y locations. 

**Describe the solution you'd like**
I would like to see those boxes for X and Y coordinates to be added to the Slicer settings in the UI when, and only when, the X seam "user specified" is selected. 

**Describe alternatives you've considered**
There are no alternatives to this, this is needed 

**Additional context**
Have a good one and thank you!

synsuiter, 2020-11-02:
Here is the applicable code (I think)

Check Highlighted sections

               "z_seam_type":
                {
                    "label": "Z Seam Alignment",
                    "description": "Starting point of each path in a layer. When paths in consecutive layers start at the same point a vertical seam may show on the print. When aligning these near a user specified location, the seam is easiest to remove. When placed randomly the inaccuracies at the paths' start will be less noticeable. When taking the shortest path the print will be quicker.",
                    "type": "enum",
                    "options":
                    {
                        "back": "User Specified",
                        "shortest": "Shortest",
                        "random": "Random",
                        "sharpest_corner": "Sharpest Corner"
                    },
                    "default_value": "sharpest_corner",
                    "limit_to_extruder": "wall_0_extruder_nr",
                    "settable_per_mesh": true
                },
                "z_seam_position":
                {
                    "label": "Z Seam Position",
                    "description": "The position near where to start printing each part in a layer.",
                    "type": "enum",
                    "options":
                    {
                        "backleft": "Back Left",
                        "back": "Back",
                        "backright": "Back Right",
                        "right": "Right",
                        "frontright": "Front Right",
                        "front": "Front",
                        "frontleft": "Front Left",
                        "left": "Left"
                    },
                    "enabled": "z_seam_type == 'back'",
                    "default_value": "back",
                    "limit_to_extruder": "wall_0_extruder_nr",
                    "settable_per_mesh": true,
                    "children":
                    {
                        **"z_seam_x":**
                        {
                            "label": "Z Seam X",
                            "description": "The X coordinate of the position near where to start printing each part in a layer.",
                            "unit": "mm",
                            "type": "float",
                            "default_value": 100.0,
                            "value": "(0 if (z_seam_position == 'frontleft' or z_seam_position == 'left' or z_seam_position == 'backleft') else machine_width / 2 if (z_seam_position == 'front' or z_seam_position == 'back') else machine_width) - (machine_width / 2 if z_seam_relative or machine_center_is_zero else 0)",
                            "enabled": "z_seam_type == 'back'",
                            "limit_to_extruder": "wall_0_extruder_nr",
                            "settable_per_mesh": true
                        },
                        **"z_seam_y":**
                        {
                            "label": "Z Seam Y",
                            "description": "The Y coordinate of the position near where to start printing each part in a layer.",
                            "unit": "mm",
                            "type": "float",
                            "default_value": 100.0,
                            "value": "(0 if (z_seam_position == 'frontleft' or z_seam_position == 'front' or z_seam_position == 'frontright') else machine_depth / 2 if (z_seam_position == 'left' or z_seam_position == 'right') else machine_depth) - (machine_depth / 2 if z_seam_relative or machine_center_is_zero else 0)",
                            "enabled": "z_seam_type == 'back'",
                            "limit_to_extruder": "wall_0_extruder_nr",
                            "settable_per_mesh": true

synsuiter, 2020-11-02:
it did not bold them...
**z_seam_x**
**z_seam_y**

marciot, 2020-11-18:
@synsuiter: Can this be closed?

synsuiter, 2020-11-23:
Thank you Sir! :D

closed 2020-11-23 by synsuiter