SynDaverCO/symple-slicer #125: Print time field culling days

SynWillinger, 2020-11-18:
**Describe the bug**
In the time estimate on the "Print and Preview" section all 24 hour increments are being dropped. For example, a print that is 25 and a half hours shows as 01:30:00, or one that is 50.25 hours shows as 02:15:00.

**Details**
 - SympleSlicer Version: 1.0.14
 - Browser: Chrome

**To Reproduce**
Steps to reproduce the behavior:
1. Slice a part that will take greater than 24 hours to print
2. Observe "Print and Preview" "Print time" field
3. Open the generated gcode to see the second line which shows the slicer estimated time in seconds
4. Divide the estimated time in seconds by 60 to get minutes, then divide by 60 again to get hours
5. Compare the values found in steps 2 and 4 and see that a multiple of 24 has been subtracted from the hours, until less than 24 remain

**Expected behavior**
Either all of the hours should be listed in the hours field, or a days field should be added

**Screenshots**
25 hour print slicer and code:
[image]
[image]

50 hour print slicer and code:
[image]
[image]

**Additional context**
I recall seeing 25:xx:xx when slicing the model I found this bug on before updating to the latest version, so I think this bug is new.

marciot, 2020-11-30:
@SynWillinger: A fix for this is now in beta (v1.0.15), close if it looks good.

SynWillinger, 2020-11-30:
Tested, looks good.

closed 2020-11-30 by SynWillinger