SynDaverCO/symple-slicer #126: Machine Settings in .toml file / Gcode flavor

stanreuben, 2020-11-29:
I have two categories

1. Gcode Flavour
2. Machine Settings .toml code

I have typed this on the .toml file [Archive.zip](https://github.com/SynDaverCO/symple-slicer/files/5612012/Archive.zip)

machine_gcode_flavor = "reprap"

Even though I have included this setting on the .toml file, still after slicing is done on MARLIN. Kindly let me know if this is an issue or if I'm doing it wrong?

I would need to slice for various firmware using Symple Slicer, I hope I can get help on this.

Secondly I would require the .toml code for Printhead Settings > X min, Y min, X max, Y max, Gantry height

Kindly help me in this regards

marciot, 2020-11-30:
@stanreuben: These are under Machine Settings in the UI:

https://syndaverco.github.io/slicer/guide/symple_slicer_users_guide.md.txt#Machine-Settings

If you wish to adjust these settings in the .toml file, they correspond to:

- machine_depth
- machine_height
- machine_width
- gantry_height

There are no settings for x_min and y_min in the slicer; these are always zero.

stanreuben, 2020-12-01:
If the slicer running backed by Cura Engine that should accept the Printhead Settings right? If there is no option in the GUI if there is anyway we can change the Printhead settings

marciot, 2020-12-02:
Yes. It is running a full Cura engine on the backend. Check the box "Show implicit values as comments" when exporting a .toml file and it will show all the available settings in the Cura engine. You can then uncomment the lines you want to change and reimport the .toml file. This will allow you to tweak settings which aren't in the GUI.

stanreuben, 2020-12-07:
I wonder whether any of this command corresponds to X min, X max, Y min, Y max of Printhead. If I'm wrong kindly correct me Since Cura Software has an option to change these, I'm pretty sure CuraEngine also will be having the command for tweaking these

`machine_extruder_end_pos_x = 0
machine_extruder_end_pos_y = 0
machine_extruder_start_pos_x = 0
machine_extruder_start_pos_y = 0
(or)
machine_head_with_fans_polygon = [[-20,10],[10,10],[10,-10],[-20,-10]]
`

Kindly assist me, I have gone through every single command

stanreuben, 2020-12-07:
[image]

This Settings X min, X max, Y min, Ymax under Printhead Settings in Cura Software, So I think the commands for Cura Engine also Should be available

marciot, 2020-12-07:
I believe those settings affect only the UI and the graphical representation of the bed. They are not passed to the Cura Engine for slicing. Anyhow, questions about the Cura or Cura Engine should be directed to Ultimaker as we do not maintain those products.

marciot, 2020-12-07:
> machine_extruder_end_pos_x = 0 machine_extruder_end_pos_y = 0 machine_extruder_start_pos_x = 0 machine_extruder_start_pos_y = 0 (or) machine_head_with_fans_polygon = [[-20,10],[10,10],[10,-10],[-20,-10]]

Where did you find this equation?

marciot, 2020-12-07:
Okay, I see that those came from "fdmextruder.def.json". These equations are indeed loaded by SympleSlicer and you'll find the corresponding values of in the toml file:

- machine_extruder_end_pos_x 
- machine_extruder_start_pos_x 
- machine_extruder_start_pos_y 
- machine_head_with_fans_polygon 

So, these values can be changed. However, I believe your question is how these values relate to the "X_min", "Y_min", "X max", "Y max" and "Gantry Height" fields in the Cura UI. I do not know. This would be a question for Ultimaker.. If you can obtain this information from them, it would be easy for us to add these fields to Symple Slicer.

marciot, 2020-12-07:
In this file, it appears as if the mapping between the fields and the Cure Engine configuration are specified as "settingKey":

https://github.com/Ultimaker/Cura/blob/abe2a56a0332c621668407356f987808c1661029/plugins/MachineSettingsAction/MachineSettingsPrinterTab.qml

Notice that the following mappings are defined:

   - Gantry Height   -> "gantry_height"
   - Number of Extruders -> "machine_extruder_count"

However, there does not appear to be a corresponding Cura Engine setting for "X min", "Y min" and "X max". Oddly, that file indicates "Y max" corresponds with "machine_head_with_fans_polygon", which seems incorrect.

To be honest, I really don't understand that source code. All this is just speculation. You would really need to follow up with Ultimaker to get more clarity on this. I am not the person to ask about this.

marciot, 2020-12-07:
@stanreuben: Okay, one more clue. Here is the corresponding Cura definition .json for the Creality CR-10:

https://github.com/Ultimaker/Cura/blob/abe2a56a0332c621668407356f987808c1661029/resources/definitions/creality_cr10.def.json

Notice that the values in the "machine_head_with_fans_polygon" corresponds more or less to the values in the "X min", "Y min", "X max" and "Y max" fields in your screenshot for the Cura UI for the Creality CR-10:

```
 "machine_head_with_fans_polygon": { "default_value": [
                [-26, 34],
                [-26, -32],
                [32, -32],
                [32, 34]
            ]
```

So, my guess is that those values are indeed populating `machine_head_with_fans_polygon` according to the following pattern:

```
 "machine_head_with_fans_polygon": { "default_value": [
                [X_MIN, Y_MAX],
                [X_MIN, Y_MIN],
                [X_MAX Y_MIN],
                [X_MAX, Y_MAX]
            ]
```

I am almost certain this is correct, but I would feel more confortable if you could check whether this relationship holds for a couple other printers beside the "Creality CR-10". If you can verify that, I would be comfortable adding those fields to the Symple Slicer UI.

stanreuben, 2020-12-08:
> @stanreuben: Okay, one more clue. Here is the corresponding Cura definition .json for the Creality CR-10:
> 
> https://github.com/Ultimaker/Cura/blob/abe2a56a0332c621668407356f987808c1661029/resources/definitions/creality_cr10.def.json
> 
> Notice that the values in the "machine_head_with_fans_polygon" correspond more or less to the values in the "X min", "Y min", "X max" and "Y max" fields in your screenshot for the Cura UI for the Creality CR-10:
> 
> ```
>  "machine_head_with_fans_polygon": { "default_value": [
>                 [-26, 34],
>                 [-26, -32],
>                 [32, -32],
>                 [32, 34]
>             ]
> ```
> 
> So, my guess is that those values are indeed populating `machine_head_with_fans_polygon` according to the following pattern:
> 
> ```
>  "machine_head_with_fans_polygon": { "default_value": [
>                 [X_MIN, Y_MAX],
>                 [X_MIN, Y_MIN],
>                 [X_MAX Y_MIN],
>                 [X_MAX, Y_MAX]
>             ]
> ```
> 
> I am almost certain this is correct, but I would feel more comfortable if you could check whether this relationship holds for a couple of other printers besides the "Creality CR-10". If you can verify that, I would be comfortable adding those fields to the Symple Slicer UI.

Yes I too thought that would be the X min, max and Y min, max because the fields were in two values with a positive and negative sign, so I thought that would be the settings I was looking for

machine_head_with_fans_polygon = [[-20,10],[10,10],[10,-10],[-20,-10]  , this is the standard X min, max and Y min, max for any fdm printer, So it is defined as default and many printers are having different X min, max and Y min max

I will attach a file of mine which I'm preparing for about 20+ printers and I got its values straight from the Companies itself.

So I'm sure if you could add these in the SImple Slicer then that would be great. 
```
Since the two which I quoted and you quoted are different in nature, the signs, Kindly have a look at it and kindly let me know about it

machine_head_with_fans_polygon = [[-20,10],[10,10],[10,-10],[-20,-10]]

"machine_head_with_fans_polygon": { "default_value": [
                [-26, 34],
                [-26, -32],
                [32, -32],
                [32, 34]
            ]
"machine_head_with_fans_polygon": { "default_value": [
                [X_MIN, Y_MAX],
                [X_MIN, Y_MIN],
                [X_MAX Y_MIN],
                [X_MAX, Y_MAX]
            ]

The Signs have changed in the Second bracket and further, Kindly check
```
[Printers UPDATED.pdf](https://github.com/SynDaverCO/symple-slicer/files/5657888/Printers.UPDATED.pdf)

marciot, 2020-12-09:
@stanreuben: My understanding is that SynDaver will not support third-party printers in our published version of Symple Slicer, however you are free to fork the project according to the terms of the license. You can reach out to "3DSupport@SynDaver.com" if you need more details.

marciot, 2021-01-04:
@stanreuben: I have added the settings you requested to the UI and to the .toml file:

[image]

I also added a corresponding "Print Sequence" option that makes use of these parameters:

[image]

In the TOML file, there are now four additional options:

      * machine_head_with_fans_x_min
      * machine_head_with_fans_y_min
      * machine_head_with_fans_x_max
      * machine_head_with_fans_y_max

These values will automatically be used to populate "machine_head_with_fans_polygon", so you should remove those from the profiles and use the four separate options instead. This will allow the settings to be editable from the UI.

You can test out the changes here:

https://syndaverco.github.io/slicer-beta/

If these changes look good and do what you need, please close this ticket. Thank you and have a happy new year!

stanreuben, 2021-01-06:
Hello there, Yes I'm now perfectly okay with everything, thank you so much I'll close the issue

closed 2021-01-06 by stanreuben