SynDaverCO/symple-slicer #130: Need to know about Profiles and Using this Open Source

stanreuben, 2021-01-06:
Hello There,

I would like to clone this project and use it. I would like to add profiles for my printers, I can see that Syncdaver and Cura Default profiles are there I would like to add other profiles as well for my Printers

I would like to use Symple Slicer on my website and I'll give attribution to SynDaver according to AGPL License

Kindly Note 

1. I would like to clone the open source project and use it through my website and will follow the AGPL License
2. I would like to add certain Profiles for my Printers
3. I would like to clone the latest beta version and would like to update it now and then

Kindly let me know in this regards and help me achieve it

stanreuben, 2021-01-07:
Kindly let me know, I'm waiting for your kind reply

Thank you

marciot, 2021-01-07:
@stanreuben: There is a new feature in beta that allows you to add profiles without forking the whole program. 

This is not necessarily a new feature, as I had the basic functionality in there already for my own testing, but now I made it visible in the UI for anyone who wants to use it. This would allow you to publish your profiles in a repo and simply direct people to add the URL to Symple Slicer; from that point on, whenever the user ran Symple Slicer it would show your profiles in addition to ours.

In order to make use of this feature, you can either run a webserver locally on your machine or publish the profiles on [github pages](https://pages.github.com/). Once the profiles are published, add the URL to the "Additional Profiles URLs" and Symple Slicer will pull them from the server.

[image]

As an example, suppose you had git hub pages with the following files:

```
/profile_list.toml
/machine_profiles/my_printer.toml
/print_profiles/my_printer.toml
```

The contents of `profile_list.toml` would be:

```
# This file maps profile filenames (excluding .toml)
# to names as it appears in the UI drop down menus

[machine_profiles]

my_printer = "My 3D Printer"

[print_profiles]

my_material = "My Print Material"
```

The contents of `machine_profiles/my_printer.toml` would be:

```
[metadata]

date = "1/1/2021"

[settings]

extruder_prime_pos_abs = true
gantry_height = 30
```

The contents of `print_profiles/my_material.toml` would be:

```
[metadata]

date = "1/1/2021"
material_notes = "Optional notes about this material"

[settings]

cool_fan_speed_max = 20
```

Forking the whole program is a more complicated affair and I can't offer any guidance on that. I suggest you read https://docs.github.com/en/free-pro-team@latest/github/getting-started-with-github/fork-a-repo to help you get started.

stanreuben, 2021-01-08:
Hey, @marciot Thank you for your kind information.

Let me tell you what and all we will be experimenting with the slicer

We will be bringing automated STL import from our website and After slicing we will save the gcode file on our web server itself, the gcode file won't be downloaded into the desktop/local. 

```
Note : 

1. STL auto import and if multiple stl also after one stl sliced next stl auto import
2. Saving the Gcode into the webserver only, not on local
3. Doing some subtle changes in CSS such as background color, model color
```

So if I fork the repository and the only changes would be saving it into the webserver, not the local, CSS changes. Will forking affect the whole process and how can I ignore forking to which we will custom change, rest all can be forked

Any ideas on this? I need some guidance on this

SynWillinger, 2021-01-09:
@stanreuben thank you for your interest, and your project sounds exciting. However, we as a company are not able to support forked projects or give specific legal advice. The project is licensed under the AGPL V3 license and is able to be forked and worked on as long as that license is followed. https://www.gnu.org/licenses/agpl-3.0.en.html

closed 2021-01-09 by SynWillinger