SynDaverCO/symple-slicer #134: output format not set if "stl" appears in filename

SynWillinger, 2021-01-14:
**Describe the bug**
If the string "stl" appears in the filename of the input model, that stl string will be replaced with "gcode" and the ".stl" will remain as ".stl". 

**Details**
 - SympleSlicer Version: 1.0.158
 - Browser: Chrome

**To Reproduce**
Steps to reproduce the behavior:
1. Load a file with a name like "*stl.stl"
2. Slice and proceed to Print and Preview
3. Observe that the "Save as" string is now "*gcode.stl"

**Expected behavior**
section before the . in the filename should not be modified, and section after the . should be changed to gcode

**Screenshots**
[image]
[image]

**Additional context**
After downloading one of the incorrectly named outputs and renaming the file to be .gcode it appears to be normal gcode output.

marciot, 2021-05-06:
@SynWillinger: This can be closed, right?

SynWillinger, 2021-05-06:
@marciot yes, I recall this one having been resolved a while ago, and just double-checked in the current release. My bad for not getting this closed sooner.

closed 2021-05-06 by SynWillinger