SynDaverCO/symple-slicer #135: Edge auto update

Rosey1855, 2021-01-21:
When I refresh the slicer in microsoft edge the slicer does not show any updates. I think the auto update could be broken.

synsuiter, 2021-03-18:
What version does your edge browser say you are on? 
Have you already tried wiping your entire browser history and cache clean?
There can be an issue if you use/have open, the beta version and the user version.

Rosey1855, 2021-03-18:
Im on version 1.0.16. I have not cleared my cache or browser history.

Mike Rosenhagen

Production Team Lead

SynDaver

3121 N Garfield Ave

Loveland, CO. 80538

Main: 970-685-4501

***@***.******@***.***>

SynDaver products have appeared on recent episodes of ABC's Shark Tank and Grey's Anatomy, Discovery Channel's MythBusters, CBS’s CSI and Sunday Morning, CNN’s Great Big Story, NBC's Crossbones and Daytime TV, PBS’s Quest, AMC's Shipping Wars as well as too many news and educational programs to list. SynDaver has also appeared in Forbes, Wired, Inc Magazine, Smithsonian, National Geographic, Popular Science, Science Magazine, The New York Post, The Huffington Post, Atlas Obscura, Today's Medical Developments, Surgical Products Magazine, Medical Product Manufacturing News, Medical Device Technology, EMS World Magazine and too many newspapers to list.

SynDaver technology is protected under U.S. Patent Numbers 10424226, 10354559, 9881522, 9679501, 9472122, 8801438, 8512044, 8425234, 8137110, 7993140, 7699615, 7677897, 7507092, 7427199 and 7272766. This message may contain confidential, proprietary, or legally privileged information and is intended only for the use of the addressee named above. No confidentiality or privilege is waived or lost by any mistransmission. If you are not the intended recipient of this message you are hereby notified that you must not use, disseminate, copy it in any form, or take any action in reliance on it. If you have received this message in error please delete it and any copies of it and notify SynDaver immediately

WARNING: This email originated from outside of SynDaver Labs. DO NOT click links or open attachments unless you recognize the sender.

synsuiter, 2021-04-09:
@Rosey1855 Have you cleared your cache and tried this again?

synsuiter, 2021-04-09:
@Rosey1855 Could this be related to #62 ?

marciot, 2021-05-07:
@synsuiter, @Rosey1855: Can this ticket be closed? I believe it refers to the old Edge. Now it is based on Chromium so I assume this is no longer relevant.