SynDaverCO/symple-slicer #140: TOML import error

synsuiter, 2021-04-01:
I was on the current, 1.0.16, version of the User Symple Slicer. 

**Describe the bug**
Not a bug, more like a communication improvement request.

I imported a .TOML file that was missing the quotes around one of the named values: "zigzag".
When I applied the profile, Symple Slicer gave me this pop up message: "Error: Cannot read property 'split' of undefined Line:"
BUT it did not give me the line that had the error, making it difficult for me to find and solve. (Thank you @marciot for finding it!)

**Expected behavior**
It should give the toml file line with the error in the pop up message. 

error producing TOML file attached as TXT file. 

[PETG_clip-board.txt](https://github.com/SynDaverCO/symple-slicer/files/6245123/PETG_clip-board.txt)

synsuiter, 2022-10-11:
does this still need to be on hold or is there a possible fix?

marciot, 2022-10-14:
@synsuiter: Right now I have not solution, as this is part of a third party code library, but maybe in the future they either release an update or I find some other way to solve the issue. I think leaving it on hold is reasonable.

synsuiter, 2022-10-14:
k, sounds like a plan.