SynDaverCO/symple-slicer #145: Firmware Update extra step request

synsuiter, 2021-04-08:
**Is your feature request related to a problem? Please describe.**
There is too high risk that a customer will have the wrong printer applied when they flash firmware through Symple slicer. 

**Describe the solution you'd like**
We would like an extra step in the firmware update process.

FOR BOTH USB AND WIFI PROCESSES:
Update Firmware -> "Update" -> **Please choose your printer** -> Continue firmware update process...

**Describe alternatives you've considered**
We have already discussed instructions and clarity on the process but too many alternatives were brought up to not apply an extra step for surety.

marciot, 2021-04-09:
This is actually tricky to do. Would a possible alternative be a yes and no question asking "You are about to flash for an SynDaver Axi?" is this correct?

synsuiter, 2021-04-09:
@marciot 

Temp Solution:

A double checking confirmation box, instead of a selection process.

Process:
- UPDATE FIRMWARE
- UPDATE
- RED WARNING BOX: "You are about to Flash your **insert applied printer name here** with **insert firmware printer and version here** Firmware. Would you like to continue?    [OK]    [CANCEL]"
- Continue update process...

synsuiter, 2021-04-14:
There has been a double check question implemented, I am going to close this ticket.

closed 2021-04-14 by synsuiter