SynDaverCO/symple-slicer #147: Add "applied printer" area

synsuiter, 2021-04-15:
**Is your feature request related to a problem? Please describe.**
If you are in mid process, there is no way of knowing which printer you currently have applied.

**Describe the solution you'd like**
somewhere static on the screen, small and in a light text, should be the type of printer that is applied to Symple slicer. 

Not sure where the best area is, it will add clutter to the screen, this has been discussed and it is felt that it is a necessary clutter.

synsuiter, 2021-04-22:
Wanted to add this:
had a tech support call yesterday where a customer was convinced the "save settings from last session" wasn't being applied properly for his machine selection.
He has an AXI 2 but claims when he tried to use his AXI 2 settings from his previous session, it was not switching to an AXI 2 but remaining an AXI 1.

I tested this and I went through the full process of slicing two prints, before with the settings applied, and after when I only kept last session settings.
The saved settings are being applied just fine, the issue was that the machine drop down defaults to AXI 1, so if they click on "load slicer settings from printer and material" to see what is applied, they do not see what is applied but instead the drop down defaults.

If we could have the "applied printer" name somewhere in the widow that just remains, it would clear up a lot of confusion.

synsuiter, 2021-04-28:
Love that placement! This does exactly what we need!

Few notes/questions from here:

1) Can we make the text smaller and change the Type Face to match the general text used elsewhere? 
2) What does it display when the program first loads up, before a customer has applied a printer for the first time? 
- This one I am going to go check in a moment but I need to clear all my browser stuff and right now I am typing up this comment. 
3) Can the text be filled in, instead of an outline? 
4) How would you feel about changing the color of the text to a darker blue but then "fading" it as a transparency so its darker but present but faded?

synsuiter, 2021-04-28:
# 2 = It displays "SynDaver Axi" before a machine has been applied.

synsuiter, 2021-04-30:
Please reduce text of printer type by ~40% we want it visible but not as prominent as it is currently.

marciot, 2021-05-06:
@synsuiter: This can be closed, right?

synsuiter, 2021-05-06:
Lookin good.

closed 2021-05-06 by synsuiter