SynDaverCO/symple-slicer #150: Moving the warning

synsuiter, 2021-05-10:
**Is your feature request related to a problem? Please describe.**
The LCD screen fade out is causing customers to panic because they are not reading the warning before seeing the fade.

**Describe the solution you'd like**
we can either make the warning pop up long before the process does that or we can make the warning more prominent. I do not think we can edit browser pops ups though, so timing is our fix.

synsuiter, 2021-05-10:
Can we have it so the firmware wipe does not begin until after "OK" is selected on that pop up?

marciot, 2021-05-10:
@synsuiter: No. According to browser security rules, once they click the "Update" button there can be no more than a second or so delay before the flash begins. If I were to put a dialog box right after they click that button, then it would violate that requirement and the flash would fail. We would have to force the user to click "Update" again in order to get another chance at flashing the firmware. They already have to click it twice to get around browser security rules, changing things as you suggest would require them to click it three times.

synsuiter, 2021-05-10:
can we make the text bigger in that warning pop up?

synsuiter, 2021-05-10:
ah, i did not think so. ok, we will discuss some other options here and get back to you on this one.

marciot, 2021-05-10:
We could also add some text in red right above the "Update" button. This text would not be part of the alert boxes. It would just always be there.

synsuiter, 2021-05-10:
Not a bad idea, I'll ask around here.

Could we have the actual stages of the button itself change? like this:
step 1) button says "ACCEPT UPDATE WARNING"
step 2) button says "BEGIN PRINTER WIPE"
step 3) button says "UPDATE PRINTER FIRMWARE"
?

marciot, 2021-05-10:
That's way too complicated to do. I think a big red warning will be sufficient. We wouldn't need three steps, just the current two.

synsuiter, 2021-05-10:
Yeah, sounds like the best option

marciot, 2021-05-10:
@synsuiter fix for this pushed to SS beta, v1.0.19

synsuiter, 2021-05-12:
I like the change, I am going to close this out.

closed 2021-05-12 by synsuiter