SynDaverCO/symple-slicer #151: Objects rearranging unexpectedly

synsuiter, 2021-05-12:
**Describe the bug**
After aranging my objects the way I want them, I decide to pick a different material, or I realize I have the wrong printer selected. Then I go any APPLY the new profile or machine.

When I do this, all models that have been arranged and placed will reload onto the bed and ignore any placement done prior to the "APPLY" of a new profile or machine. 

**Expected behavior**
If possible, we would like the arranged objects to stay in the same position they were placed in when "APPLY"ing a new profiles.

marciot, 2021-05-12:
I can look into this, but changing a printer is sort of like taking the table cloth off a table. Everything comes off with it...

synsuiter, 2021-05-12:
Darn, That's kind of what I was thinking was happening, we can close this as a "wont fix" if there is nothing to be done.

closed 2022-10-20 by synsuiter