SynDaverCO/symple-slicer #152: Special Modes: One at a Time

synsuiter, 2021-05-12:
**Describe the bug**
Feature not working/broken.

**Details**
 - SympleSlicer Version: 1.0.19 (BETA)
 - Browser [e.g. chrome, safari]: Chrome

**To Reproduce**
Steps to reproduce the behavior:
1. Open Symple Slicer
2. Apply profile and machine settings
3. In "slice settings", under "Special Modes" change the Print Sequence to "One at a time"
4. Slice.
5. See resulting preview, or try running the print. Both options still have all models, regardless of placement or number, printing at the same time.

**Expected behavior**
They should be printing one at a time or giving the error that parts are inaccurately placed for the needs of the tool head and that mode setting. 

**Screenshots**
[image]
[image]

synsuiter, 2021-05-12:
I was able to confirm that the setting is being applied.

I applied a PLA and AXI 2 profile
Then I changed the Print sequence option in the Slice Settings to "One at a time"
Then I saved my print settings.
The option was present and active: **print_sequence = "one_at_a_time"**

synsuiter, 2021-05-12:
Cura Ulti Function: 

[image]
[image]
[image]
[image]

synsuiter, 2021-05-12:
I edited the Printhead settings and Gantry in CURA Ulti and when I selected "One at a time" it altered the build volume accessibility. This is normal and should happen. 
[image]

marciot, 2021-06-07:
@synsuiter: So I contacted Ultimaker about this and they informed me that printing one at a time is not a Cura Engine feature. It is actually something that is implemented in the Cura UI itself. There are a lot of roadblocks to bringing this functionality in Symple Slicer and doing so would be a major development effort. Roughly, the problem consists of two parts:

1. Validating whether the print can happen without tool head collisions
2. Actually slicing the file and generating GCODE

While 2 is moderately difficult and could be done by running the slicer multiple times and then combining the resulting GCODE, 1 is extremely difficult to do correctly and I don't think Symple Slicer could pull it off.

So, assuming we really need this feature, we probably would have to place the onus on the user to make sure the parts were sufficiently separated and there would be no warning if this were not the case,

I feel that maybe this feature isn't important enough for a half-assed implementation and that we probably ought to remove the option entirely from Symple Slicer and recommend users use Ultimaker Cura if they want this feature.

synsuiter, 2021-06-07:
That is a fair, darn. Okydoky, I'll bring this up today and let everyone know.

Thanks for checking into it @marciot

marciot, 2021-09-02:
@synsuiter: As of the latest SS beta, one-at-a-time mode has been enabled.

It turns out either Ultimaker misled me about the CuraEngine not supporting this, or I misinterpreted what they said; it turns out that the engine can indeed do one-at-a-time mode, but it required some tweaking to the way SS presents STL files to the engine. With that change done, it seems to work reasonably well.

There is still no validation for print collisions, so I would consider this to be a feature for advanced users that know how to arrange objects on the print bed without interference.

synsuiter, 2022-10-20:
This does appear to be working normally, I agree, advanced it is.

closed 2022-10-20 by synsuiter