SynDaverCO/symple-slicer #156: Beta Supports preventing a complete slice

synsuiter, 2022-03-16:
**Describe the bug**
When Supports are enabled in the Beta Symple slicer, the slice never "finishes" it only remains "slicing" (loading). 

**Details**
 - SympleSlicer Version: 1.1.0
 - Browser: google Chrome

**To Reproduce**
Load any model into symple slicer beta
enable supports is settings
Slice.

Please note: Beta Slices just fine if supports are NOT enabled.

Here are **Screenshots** of the browser logs during a fail and the slicer logs.
[image]
[image]

marciot, 2022-03-18:
@synsuiter: I've seen this error before. This is happening within the Cura engine, so there is nothing I can do in Symple Slicer to address the issue. However, it seems to be caused by a profile setting since if you slice with the Cura defaults, this does not happen. The only way to track this down would be to start with the Cura defaults and then merge in bits from the desired profile until the crash manifests itself. Then narrow it down to the particular option, or combination of options that causes the crash.

marciot, 2022-03-18:
@synsuiter I found an STL that exhibited the problem and determined that downgrading to CuraEngine 4.12.1, the prior release, solved the problem. Let's give it a try and see if it is better.

synsuiter, 2022-03-21:
Working well for me, closing this ticket.

closed 2022-03-21 by synsuiter