SynDaverCO/symple-slicer #159: wifi printing error: addr is not defined

SynWillinger, 2022-05-02:
**Describe the bug**
In the current beta, if I try to send a sliced gcode to a machine, I get the following error:
[image]

**Details**
 - SympleSlicer Version: beta 1.1.0 as of 5/2
 - Browser: Chrome

**To Reproduce**
Steps to reproduce the behavior:
1. Set up a machine in the Wifi Printing section
2. Slice a gcode
3. Select "send gcode to printer"
4. See error

**Expected behavior**
no error should be encountered, and the gcode should be sent to the machine

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
The same error is encountered if you try to update wireless firmware. No error is seen but nothing happens if I try to use "manage" from the "Wireless Printing" section

synsuiter, 2022-05-02:
confirmed on my end
[image]

SynWillinger, 2022-05-03:
This issue has been corrected based on our testing.

closed 2022-05-03 by SynWillinger