SynDaverCO/symple-slicer #160: Note added to 1st slice Setting box

synsuiter, 2022-05-02:
**Is your feature request related to a problem? Please describe.**
To a previous user, it is not clear during the normal process that they can change their setting options.

**Describe the solution you'd like**
I would like to add this note to the "settings" section when 1st slice is active. 

"To access more setting options, go to 'Advanced Features' and customize your 'User Interface' experience.

**Describe alternatives you've considered**
No other alternate considered, this seems the simplest way to communicate to those who wont know what to do when they see no setting options.

marciot, 2022-05-04:
@synsuiter: First slice is only active if the user changes to that mode, so at that point they would already know how to change mode. The "syndaver-default" is the mode that comes up by default. Do you want first slice to be the default mode?

synsuiter, 2022-05-04:
oh! ok, when I loaded mine up it was in first print mode, that must have been from a previous load tho. ok, no worries. 

I have some confusion on how 1st print mode would work because if it was their first print, then they wouldnt know to go set it to "1st print" mode so they would just be using default. II dont know the best solution here, maybe we can discuss it in the meeting.

synsuiter, 2022-05-06:
I think we are good here now.

closed 2022-05-06 by synsuiter