SynDaverCO/symple-slicer #170: No support option for beginners

synsuiter, 2022-08-18:
**Describe the bug**
The "Beginners" setting level option has a "Support" section but does not show the Check box when expanded. It is effectively empty. 

**Details**
 - SympleSlicer Version: 1.1.1 BETA
 - Browser : Chrome

**To Reproduce**
Steps to reproduce the behavior:
1. open Symple Slicer
2. Go to the Advanced settings
3. Apply the setting level "Beginner" 
4. Select and apply a printer and a material
5. Go to the Slicing section and try to turn on the supports. 

**Expected behavior**
There should be the option to turn the support on and once turned on, no other settings will appear.
Simply and "ON" - "OFF" option. 

**Screenshots**
[image]

marciot, 2022-08-19:
@synsuiter: This is because "support_enable" was missing from the beginner settings. I added it.

synsuiter, 2022-08-19:
Gracias, closing ticket.

closed 2022-08-19 by synsuiter