SynDaverCO/symple-slicer #171: Models moving when sliced

synsuiter, 2022-08-18:
**Describe the bug**
When slicing more than one unique model, they will reposition themselves when you engage the "slice" function. 
This was confirmed with numerous model pairs, they only need to be different models. 

**Details**
 - SympleSlicer Version: 1.1.1 BETA
 - Browser: Chrome

**To Reproduce**
Steps to reproduce the behavior:
1. Open Symple Slicer
2. Select an AXI printer and any material profile and APPLY.
3. Load in at least 2 models that are NOT the same. 
4. Position them for slicing.
5. Slice those models
6. Observe that the slice does not match with where you positioned your models.

**Expected behavior**
The models should stay positioned where they were intended by the user. 

**Screenshots**
BEFORE SLICE
[image]
[image]

AFTER SLICE
[image]
[image]

marciot, 2022-08-19:
@symsuiter: Have been unable to reproduce this.

marciot, 2022-08-19:
@symsuiter: Export your settings and make sure nether mesh_rotation_matrix, mesh_position_x, mesh_position_y nor mesh_position_z are in the settings. If it is, it may override the positioning for objects.

synsuiter, 2022-08-22:
Upon removing the following settings as a factor, the models still moved at time of slice. 

- mesh_position_x
- mesh_position_y
- mesh_positionz
- mesh_rotation_matrix

synsuiter, 2022-08-22:
When I change all those settings to a value of "0", there is no change in the shifted position result when sliced.

marciot, 2022-08-22:
@synsuiter: You should not change the values to zero. You should remove them altogether.

synsuiter, 2022-08-22:
I did both, I was reporting that removing them AND changing them, make no difference.

synsuiter, 2022-08-22:
Here are the profiles I have tested. 
[offset-tester.txt](https://github.com/SynDaverCO/symple-slicer/files/9396212/offset-tester.txt)
[offset-tester_changed.txt](https://github.com/SynDaverCO/symple-slicer/files/9396213/offset-tester_changed.txt)
[offset-tester_fixed.txt](https://github.com/SynDaverCO/symple-slicer/files/9396214/offset-tester_fixed.txt)
[offset-tester_removed.txt](https://github.com/SynDaverCO/symple-slicer/files/9396215/offset-tester_removed.txt)

synsuiter, 2022-08-22:
Fixed!! :D closing ticket.

closed 2022-08-22 by synsuiter